0xchocolate/flipperzero-wifi-marauder #1: Add console logging to marauder companion

pacmandu, 2022-09-23:
### Description of the feature you're suggesting.

Hi there! I was wondering if console logging to a text file on the flipper could be added to the Wifi Marauder Companion plugin? That way any data data captured like the PMKID is saved and full logs are backed up to the SD card without needing to solder to the dev board.

GRIT1085, 2023-01-17:
Hello, how would I go about viewing the packets received while performing an attack - deauth? I can easily perform the function, however there is no where to find information on what was gathered it seems.

0xchocolate, 2023-01-17:
Hi, the packets would be saved to the SD card on the wifi board (not the flipper), provided that you've installed one.

0xchocolate, 2023-01-17:
https://github.com/justcallmekoko/ESP32Marauder/wiki/flipper-zero#sd-card-modification

0xchocolate, 2023-03-25:
Fully supported (saving both pcaps and logs) in latest release.

closed 2023-03-25 by 0xchocolate