0xchocolate/flipperzero-wifi-marauder #5: Evil Twin attacks with the ESP32?

sealldeveloper, 2023-01-20:
### Description of the feature you're suggesting.

This would be a helpful feature to utilize.

### Anything else?

Love the program!

0xchocolate, 2023-02-04:
Is this a request for marauder FW? You might want to file it on that project instead. Please also describe exactly what you want the feature to do.

sealldeveloper, 2023-02-05:
Oh, apologies.

The feature is for an SSID to be replicated and created as a new network, and then to simultaneously deauth attack the network with the same name.

Eg.
Deauth a WiFi with the SSID "Example Wifi"
Create a new, passwordless network with the SSID "Example Wifi"
Devices would connect to the new created network instead of the real "Example Wifi"

sealldeveloper, 2023-02-05:
https://wikiless.org/wiki/Evil_twin_(wireless_networks)

closed 2023-03-17 by 0xchocolate