0xchocolate/flipperzero-wifi-marauder #6: Save marauder pcap's to flipper sdcard

fOmey, 2023-02-10:
### Description of the feature you're suggesting.

Instead of adding an additional sdcard slot to the breakout board could we potentially save pcaps's to the flipper's sdcard?

tcpassos, 2023-03-01:
I opened a PR to the official repository with this implementation. Waiting now.

fOmey, 2023-03-01:
> I opened a PR to the official repository with this implementation. Waiting now.

Very cool, can you go into a little detail on how it works? What additional pins are required?

tcpassos, 2023-03-01:
Both the wifi devboard and the flipper have an additional serial channel (Low Power UART), which serves for data transmission.
The PR I submitted to the Marauder firmware repo allows the wifi board to send packets through serial instead of SD card. And the PR I made for that repo allows the app on the flipper to listen to these packets and write them to the internal SD card.

If you are using the original wifi devboard, no hardware modification is needed, just snap the board into the flipper. If it is another wifi card, you would have to confirm that the LP_TX/LP_RX pins are properly connected to the flipper's LP_RX/LP_TX inputs.

tcpassos, 2023-03-14:
> I see that the compiled .fap was added into RM firmware, however after desoldering the shifting sd-card i had attached on the dev board, i can't find any pcap files saved anywhere on the sd-card. What modifications are required?

Hello. Are you sure you wanted to post this question here?

Today, no custom firmware has the ability to store PCAPs directly into the flipper's internal memory. I opened this PR for this implementation to be officially added. I also opened a PR in the Marauder firmware repository to send the captured packets via serial:
https://github.com/justcallmekoko/ESP32Marauder/pull/233

Both PR need to be accepted for this functionality to become official.

If you want, you can manually build the FAP and Marauder firmware that are in my forks and start using it.

tcpassos, 2023-03-15:
> I've installed the esp32_marauder_no_sd_20230301.bin in to my wifi devboard but it hasn't saved scanned eapols to the flipper zero sd card. Do I need to install an additional firmware for the flipper for this to work?

You need the modified FAP file to receive packets through serial:
https://github.com/tcpassos/flipperzero-firmware-with-wifi-marauder-companion/releases

The latest version of RogueMaster already incorporated this implementation.

closed 2023-03-17 by 0xchocolate