0xchocolate/flipperzero-wifi-marauder #10: No comma and dash on the keyboard (multi-select and add parameters problem)

LowSkillDeveloper, 2023-04-14:
### Describe the enhancement you're suggesting.

In wifi marauder, after scanning, we select an access point using the "select -a" command. 
For example, after scanning, I have "list -a" in which there are 40 access points. And I want to pick a few. 
Wifi murader supports multi-selection, you can write several numbers separated by commas like this "select -a 3,6,8,15,19,22,23" and all these points will be selected, you will not need to open the "select" item in menu every time and enter new number.

But the problem is that in the keyboard that is used in the application, as I understand it is a stock keyboard, there is no comma, there is also no dash, to add a parameter, for example, choosing "Sniff PKMID on channel", I want to add the -d parameter,  so that it would be like this "sniffpkmid -c 11 -d", so that deauthentication is also performed.  Also no colon to enter mac address

Perhaps you can somehow modifie keyboard in the application?

tcpassos, 2023-04-16:
I'm not sure if in other FW it's like this. But at least in xtreme the keyboard has a key in the lower left corner, similar to an "A+" that changes the keyboard layout and there appear the comma and the dash. Another tip is that to write the space badta hold the underscore (_).

0xchocolate, 2023-04-16:
I would consider this low priority because there are workarounds and this seems non-trivial to implement on the app side. It would be better to implement the keyboard changes on the fw side (or if there already is a better one we can use, figure out how to include it).

For the select example, you shouldn't have to keep opening the select item in the menu. You can type in your number, send it, and then when you press back, it should prompt you for another parameter. (Just backspace the previous number, type in the new one, send it, and repeat.)

For your sniffpmkid example, you can use the ForcePMKID setting if that's not already enabled: https://github.com/justcallmekoko/ESP32Marauder/wiki/settings-cmd

closed 2023-05-02 by 0xchocolate