0xchocolate/flipperzero-wifi-marauder #14: [bug] Cannot scroll the top of the list when scan list has large number of entries

xologram, 2023-05-14:
### Describe the bug.

when i run scan ap i get 150+ networks. if i go to list to select an ap the listing starts at number ~70. and i cannot scroll up to to see scanned aps 1-70. 

### Reproduction

1. run scan ap in a metropolitan area with large number of wifis
2. scan 150+ wifis 
3. list aps
4. verify that the list starts at around number 70 and aps 1-60 are not shown

0xchocolate, 2023-05-16:
Did you enable logs saving to sd card? If that's turned off, then the buffer isn't big enough to hold all the entries.

xologram, 2023-05-16:
Yep it logs to SD card and I can find list.log with all 150 entries in it will start showing the list in the companion app from 70th entry onwards.

0xchocolate, 2023-05-17:
Did you try using the menu entry "View Log > from start"? It's at the top of the main menu in the app.

xologram, 2023-05-17:
Ohh I see, I guess I was using it wrong. I was using the "List" option from the menu. But if I view the log from start (after I've click the List) it actually works and paginates properly. 

Thanks! I guess we can close this.

closed 2023-05-17 by xologram