0xchocolate/flipperzero-wifi-marauder #21: How to Edit ? (if possible)

Dieudumonde, 2023-06-25:
Hey,

I just wanted if it was possible to edit the ESP32_WiFi_Marauder.fap file to change the rickroll by anything else ?

0xchocolate, 2023-06-25:
Watch this video and pay attention to how to execute a beacon attack with an ssid list: https://youtu.be/YpVu-UF6k10

If that doesn't help, look through the marauder wiki or ask on discord.

Dieudumonde, 2023-06-26:
thank you ! but i don't have all the same options like in the video, i don't have the scripts option for exemple, i just have to update the marauder or i have to add it by myself ?

0xchocolate, 2023-06-26:
Update the app

closed 2023-06-26 by 0xchocolate

Dieudumonde, 2023-06-26:
where is the new ESP32_Wifi_Marauder.fap file ? because the "problem" womes from this file i think