0xchocolate/flipperzero-wifi-marauder #28: Unsure where to place Evil Portal index.html and ap.config.txt

SunnySkye, 2023-09-04:
Hi there
This could just be me being a bit dense.
But I when I go to use the Evil Portal function in the ESP32 Marauder application, it says that it can't find /index.html.

[image]

I have sample files ready to go, but I am just not sure where to place them.

bitterbuick, 2023-09-06:
Check out the instructions on the fork here: https://github.com/justcallmekoko/ESP32Marauder/wiki/evil-portal-workflow

Hillehom, 2023-09-13:
- nazwa: Zbuduj pakiet aplikacji płetwy ( .fap )
                wykorzystuje: flipperdevices / flipperzero-ufbt-action @ v0.1.2

opskovitch, 2023-09-16:
I am encountering the same issue and noticed I initially missed that important note in docs: https://github.com/justcallmekoko/ESP32Marauder/wiki/evil-portal-workflow#important-note

Not very experienced on the topic but it looks like 
- `/apps_data/marauder/ap.config.txt` is read from the Flipper Zero SD Card 
- `/index.html` is read from the root of the ESP32 SD Card, which requires extra hardware enabling the ESP32 SD card capabilities, using either a [SD adapter](https://www.tindie.com/products/justcallmekoko/wifi-dev-board-micro-sd-adapter/) board or a [MicroSD Breakout](https://www.sparkfun.com/products/544) according to the following instructions: https://github.com/justcallmekoko/ESP32Marauder/wiki/flipper-zero#sd-card-modification

@bitterbuick did I get this right ?

Sonkeiseven, 2023-11-02:
Bonjour, j'ai réussi en suivant cette vidéo : https://www.youtube.com/watch?v=ktIvzB__pTs

et ce github : https://github.com/bigbrodude6119/flipper-zero-evil-portal#erasing-firmware

ssshake, 2023-11-28:
I think the confusing here might be the same I'm encountering. I just flashed the Marauder firmware using the online tool (https://fzeeflasher.github.io/).

Then I installed this application through the flipper labs website. 

Inside of this app there is an option to start evil portal. It just seems to hang. 

Some documentation says to put index.html and ap.config.txt in the root of the sd card on the ESP32 device (mine does not have an sd card slot, it's the official flipper wifi dev board).

Other instructions say to put these files into /app_datas/evil_portal (which didn't exist).

Neither work.

It seems that evil portal can be its own stand alone firmware, or baked into marauder. 

It seems that there may be a stand alone evil portal app?(confused).

So it's just confusing which instructions to refer to.

Can the Marauder firmware, on a wifi dev board with no SD card, access index.html and ap.config.txt from the SD card on the flipper itself? If so where and how?

Selecting Update SD in this app's interface says SD support is disabled. Not sure what to make of it. Any help would be appreciated.

0xchocolate, 2023-11-28:
As of Marauder FW v0.13.5 and app version v0.6.5:
If your esp32 board doesn't have an sd card, then use the new option in the app to load HTML to Evil Portal. It will open a file browser for you to select the file from the flipper sd card, then it will send the file contents to the esp32. Make sure you also have an ap and/or ssid selected from their respective lists (you can add your own ssid by name). Then start the portal.

opskovitch, 2023-11-28:
@0xchocolate amazing thanks

0xchocolate, 2023-12-03:
Please provide the following logs:

1. Boot log from starting up (to verify the version)
2. After loading html
3. After starting Evil Portal

@ATKellett04

ATKellett04, 2023-12-03:
Apologies, unsure about getting a boot log but here are the others
[evilportal_0.log](https://github.com/0xchocolate/flipperzero-wifi-marauder/files/13540097/evilportal_0.log)
[evilportal_1.log](https://github.com/0xchocolate/flipperzero-wifi-marauder/files/13540101/evilportal_1.log)

Thanks

0xchocolate, 2023-12-03:
Is this a published portal (known to work) or your own file? Could you provide the html file?

0xchocolate, 2023-12-03:
Also, does it always fail? Have you tried retrying and/or resetting?

ATKellett04, 2023-12-03:
Yeah its a public html but i've tweak a couple things but only text so it has no reason not to work, and yeah Ive reset the board a couple times, i've even reloaded the companion app and the firmware and always seems to fail :/ being a pain in my bum

m00nyONE, 2024-01-26:
> As of Marauder FW v0.13.5 and app version v0.6.5: If your esp32 board doesn't have an sd card, then use the new option in the app to load HTML to Evil Portal. It will open a file browser for you to [...]

Is there a way to use this as a script? I can't get it to work sadly.

i have a script like this:
`
ssid -a -n TestPortal
evilportal -c sethtml portals/testportal.html
evilportal -c start
`

the template works and is tested.
But it seems like the upload of the html is via serial? so how do i send it over as a command ?

closed 2024-02-07 by 0xchocolate