0xchocolate/flipperzero-wifi-marauder #30: PCAP files are empty

aamkye, 2023-09-26:
Hi, after flashing the latest version (without `_sd_serial`) as it does not support microSD, I tried to capture beacons in the `Sniff->Beacon` option, but the PCAP file is empty. I was using v0.12.0 FW and MC v0.6.2.

Also, I've tried to capture the `Sniff -> RAW` but without success, the PCAP file was still empty.

aamkye, 2023-09-26:
On the other hand, flashing firmware with `_sd_serial` generates PCAP files, but all values are NULL.

aamkye, 2023-09-26:
BTW, I'm using [Xtreme-Firmware](https://github.com/Flipper-XFW/Xtreme-Firmware) version `XFW-0051_01092023`.

dikei100, 2023-10-16:
Hi aamkye,

which WiFi-board are you using? Had the same issue with my self-build board but was able to solve it. Maybe I can help you.

LowSkillDeveloper, 2023-12-13:
I update firmware of my flipper and update companion to 0.6.5 and Flipper Zero WiFi Dev Board to 0.13.5. Flash esp32_marauder_v0_13_5_20231126_flipper_sd_serial.bin, and now i have this problem...  Before this, everything worked, but I don’t remember what version it was, but it was old when they just started adding scripts. All pcap files have 0 size...

LowSkillDeveloper, 2023-12-14:
> I update firmware of my flipper and update companion to 0.6.5 and Flipper Zero WiFi Dev Board to 0.13.5. Flash esp32_marauder_v0_13_5_20231126_flipper_sd_serial.bin, and now i have this problem... [...]

I solved the problem by simply downloading the latest build version from [here](https://github.com/0xchocolate/flipperzero-wifi-marauder/actions/workflows/build.yml)...

closed 2023-12-16 by 0xchocolate