0xchocolate/flipperzero-wifi-marauder #31: I am encountering the exact same issue, and tried copying the `index.html` file at various locations on the Flipper Micro SD card, at:

Hillehom, 2023-09-26:
I am encountering the exact same issue, and tried copying the `index.html` file at various locations on the Flipper Micro SD card, at:
- /index.html
- /apps_data/index.html
- /apps_data/marauder/index.html

None of these seem to work, error persists.

Then I noticed I initially missed an important note in docs: https://github.com/justcallmekoko/ESP32Marauder/wiki/evil-portal-workflow#important-note

Not very experienced on the topic but could it be that `/apps_data/marauder/ap.config.txt` is read from the Flipper Zero SD Card, while `/index.html` is read from the root of the ESP32 SD Card ? 

Note: That would require extra hardware enabling the ESP32 SD card capabilities, using either a [SD adapter](https://www.tindie.com/products/justcallmekoko/wifi-dev-board-micro-sd-adapter/) board or a [MicroSD Breakout](https://www.sparkfun.com/products/544) according to the following instructions: https://github.com/justcallmekoko/ESP32Marauder/wiki/flipper-zero#sd-card-modification

@bitterbuick did I get this right ?

_Originally posted by @opskovitch in https://github.com/0xchocolate/flipperzero-wifi-marauder/issues/28#issuecomment-1722240238_

opskovitch, 2023-09-26:
@Hillehom I wouldn't be surprised is this issue get closed by maintainers, since 

- it's a copy-paste of a comment on an already open issue on the repo. 
- issue title is not self-describing 

Would you mind posting the problems you're encountering on the open issue ( https://github.com/0xchocolate/flipperzero-wifi-marauder/issues/28 ) and removing that one ? 
Let's join efforts and make maintainers happy :)

Hillehom, 2023-10-08:
> ```
>           I am encountering the exact same issue, and tried copying the `index.html` file at various locations on the Flipper Micro SD card, at:
> ```
> 
> * /index.html
> * /apps_data/index.html
> * /apps_data/marauder/index.html
> 
> None of these seem to work, error persists.
> 
> Then I noticed I initially missed an important note in docs: https://github.com/justcallmekoko/ESP32Marauder/wiki/evil-portal-workflow#important-note
> 
> Not very experienced on the topic but could it be that `/apps_data/marauder/ap.config.txt` is read from the Flipper Zero SD Card, while `/index.html` is read from the root of the ESP32 SD Card ?
> 
> Note: That would require extra hardware enabling the ESP32 SD card capabilities, using either a [SD adapter](https://www.tindie.com/products/justcallmekoko/wifi-dev-board-micro-sd-adapter/) board or a [MicroSD Breakout](https://www.sparkfun.com/products/544) according to the following instructions: https://github.com/justcallmekoko/ESP32Marauder/wiki/flipper-zero#sd-card-modification
> 
> @bitterbuick did I get this right ?
> 
> _Originally posted by @opskovitch in [#28 (comment)](https://github.com/0xchocolate/flipperzero-wifi-marauder/issues/28#issuecomment-1722240238)_

Hillehom, 2023-10-08:
https://github.com/0xchocolate/flipperzero-wifi-marauder/issues/31#issue-1913935184

closed 2024-02-07 by 0xchocolate