0xchocolate/flipperzero-wifi-marauder #32: Scripts documentation is absent

RekordNadoev, 2023-10-09:
The App can use some kind of scripts.

From v0.3.4 changelog:
" Scripts are here! Thanks to [@tcpassos 2]
...
The scripts are saved in the “apps_data/marauder/scripts” folder as JSON files."

Where could one find the documentation or examples of the scripts?

cyptus, 2023-10-30:
find the syntax overhere:
https://github.com/0xchocolate/flipperzero-wifi-marauder/blob/a7902aade81c71ced821a30c7e864bcb3578a82b/applications/external/wifi_marauder_companion/script/wifi_marauder_script.h

```json
{
     "meta": {
         "description": "My script",
         "repeat": times the script will repeat (default 1),
         "enableLed": true (default) | false,
         "savePcap": true (default) | false
     },
     "stages": {
         "scan": {
             "type": "ap" | "station",
             "timeout": seconds,
             "channel": 1-11
         },
         "select": {
             "type": "ap" | "station" | "ssid",
             "filter": "all" | "contains -f '{SSID fragment}' or equals '{SSID}' or ...",
             "indexes": [0, 1, 2, 3...],
         },
         "deauth": {
             "timeout": seconds
         },
         "probe": {
             "timeout": seconds
         },
         "sniffRaw": {
             "timeout": seconds
         },
         "sniffBeacon": {
             "timeout": seconds
         },
         "sniffDeauth": {
             "timeout": seconds
         },
         "sniffEsp": {
             "timeout": seconds
         },
         "sniffPmkid": {
             "forceDeauth": true (default) | false,
             "channel": 1-11,
             "timeout": seconds
         },
         "sniffPwn": {
             "timeout": seconds
         },
         "beaconList": {
             "ssids": [
                 "SSID 1",
                 "SSID 2",
                 "SSID 3"
             ],
             "generate": number of random SSIDs that will be generated,
             "timeout": seconds
         }
         "beaconAp": {
             "timeout": seconds
         }
         "exec": {
             "command": Command (eg: "clearlist -a")
         }
         "delay": {
             "timeout": seconds
         }
     }
 }
 
 Note: It is possible to inform "stages" as an array, allowing ordering and repetition of stages of the same type:
     "stages": [
       {
         "beaconList": { "ssids": ["SSID 1", "SSID 2"] }
       },
       {
         "beaconList": { "generate": 4 }
       },
     ]
}
```

RekordNadoev, 2023-10-31:
Thank You for your help and for the app!

m00nyONE, 2024-01-26:
Is there a way to evil portal html templates in a script? I can't get it to work sadly.

i have a script like this:
```
{
	"meta":	{
		"description":	"EvilPortal-McDonalds",
		"repeat":	1.0000000000000000
	},
	"stages": [
		{
			"exec":	{
				"command":	"ssid -a -n TestPortal"
			}
		},
		{
			"exec":	{
				"command":	"evilportal -c sethtml portals/TestPortal.html"
			}
		},
		{
			"exec":	{
				"command":	"evilportal -c start"
			}
		}
	]
}
```

the template works and is tested.
the html file is located under `/apps_data/marauder/portals/TestPortal.html`
But it seems like the upload of the html is via serial? so how do i send it over as a command ?

0xchocolate, 2024-01-26:
Change the sethtml command to sethtmlstr as you noted, then the next "command" should be the full html contents. (I don't know whether this would work in a script, but try it out!)

m00nyONE, 2024-01-26:
I'll give it a try. Would be nice if parsing a file could be a feature in code. Shouldn't be that hard to parse it too.
something like `$file{portals/TestPortal.html}`.

so the parser would look for $file and load the file in brackets as a string. ( as some sort of function )

I think that would be a very nice addition and open up some more possibilities.
This is also expandable for future use of such escape sequences like $exec or $time etc ...

closed 2026-11-02 by 0xchocolate