0xchocolate/flipperzero-wifi-marauder #46: Wifi DevBoard installation does not work

tuxsaporo, 2023-12-12:
I'm trying to install the Marauder firmware on my Flipper dev board, and even after the binary has been flashed, a red light appears after boot.
I tried to install it several times, but there was no success.

0xchocolate, 2023-12-12:
How are you flashing it? (Which process are you using?)

Have you tried any of the app functions? Is this an issue with this app?

tuxsaporo, 2023-12-12:
I've tried using the FZEaztMarauderFlash.
Even after the reset button is pressed, I see the red light.

[image]

https://github.com/0xchocolate/flipperzero-wifi-marauder/assets/100784790/b77efca9-97f0-44d8-bced-aae13424f68d

0xchocolate, 2023-12-12:
Okay? What happens if you try to scan?

tuxsaporo, 2023-12-12:
Nothing. 
It's like the ESP cannot boot the firmware.

0xchocolate, 2023-12-12:
That LED sequence (blue -> green -> red) is indeed marauder fw booting up.

I don't know what you're having trouble with.

tuxsaporo, 2023-12-12:
I'll try again.
Maybe there is a problem with my board.
Thank you so much.

closed 2023-12-12 by tuxsaporo

0xchocolate, 2023-12-12:
I don't think there's a problem with your board - it's clearly flashed...