0xchocolate/flipperzero-wifi-marauder #48: Button Order - Attack, Sniff

vollukas, 2024-01-04:
It would be very practical to have Sniff button right bellow Attack button because for EAPOL you need to death and then sniff raw...

Would it be possible to add this change to new release or could you point me out where should I change the source code and rebuild it myself? Thanks for the response!

0xchocolate, 2024-01-04:
Why not use sniff pmkid? Or if you really want the raw traffic, then create a script in the app for your workflow instead of manually doing it.

I'm eventually redoing the menus and will consider this feedback.

vollukas, 2024-01-04:
Active PMKID sniff only captures first EAPOL :/ I'm not skilled with PMKID so I'm also not sure what passive PMKID captures in my case.

Thank you very much for response, I will read more documentation and making a script will solve all my problems! Also looking forwards for future updates!

closed 2026-11-02 by 0xchocolate