0xchocolate/flipperzero-wifi-marauder #60: Error: Outdated App

a1exus, 2024-09-10:
Hello,

I grabbed [latest](https://github.com/0xchocolate/flipperzero-wifi-marauder/actions/workflows/build.yml) , copy it to `SD Card/apps/GPIO`, yet still getting the error

> Error: Outdated App

using https://github.com/0xchocolate/flipperzero-wifi-marauder/actions/runs/10784527304 now

0xchocolate, 2024-09-10:
On which flipper FW version?

a1exus, 2024-09-11:
> On which flipper FW version?

latest (5F4F4FCC)

0xchocolate, 2024-09-11:
Did you try grabbing the corresponding app build: flipperzero-wifi-marauder-f7-api-73.0-5f4f4fcc

a1exus, 2024-09-11:
> Did you try grabbing the corresponding app build: flipperzero-wifi-marauder-f7-api-73.0-5f4f4fcc

i obviously did not ;p, thank you for point this out for me))

thanks!

closed 2024-09-11 by a1exus