0xchocolate/flipperzero-wifi-marauder #62: Help Flipper Zero Multiboard Expansion (AIO BOARD V1.4)

H4ckD4d, 2024-11-18:
Hello everyone,

I have a Flipper Zero Multiboard Expansion Board (AIO BOARD V1.4), which includes Wi-Fi, NRF24, ESP32, a 2.4G module, NRF Mousejacker, a high-gain CC1101 module, and an SD card slot. Unfortunately, I can’t get it to work properly.

Could someone point me to any resources or documentation that explain how to set it up? Specifically, I’d like to understand:
	1.	What steps are required to initialize and configure this board?
	2.	What role the SD card plays and how it should be formatted or used?

Any advice or direction would be greatly appreciated. Thank you in advance for your help!

DeanZilla, 2025-01-03:
I have the same board, It isn't flashed, I've tried following 
[FZEasyMarauderFlash](https://github.com/SkeletonMan03/FZEasyMarauderFlash)

But I get the following:

Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
Checking for serial port...
You are most likely using an DrB0rk S3 Multiboard
Erasing firmware...
esptool.py v4.8.1
Serial port /dev/cu.usbmodemflip_Ami7orth1
Connecting......................................
Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Waiting 5 seconds and trying again...
Erasing firmware...
esptool.py v4.8.1
Serial port /dev/cu.usbmodemflip_Ami7orth1
Connecting......................................
Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Waiting 5 seconds and trying again...
Erasing firmware...
esptool.py v4.8.1
Serial port /dev/cu.usbmodemflip_Ami7orth1

Neo2a, 2025-01-04:
Need these files. Use the floowing instruction: https://www.youtube.com/watch?v=Mnv05jEKFvA
[esp32_marauder AIO board.zip](https://github.com/user-attachments/files/18305939/esp32_marauder.AIO.board.zip)

x01rrr, 2025-01-06:
> Need these files. Use the floowing instruction: https://www.youtube.com/watch?v=Mnv05jEKFvA [esp32_marauder AIO board.zip](https://github.com/user-attachments/files/18305939/esp32_marauder.AIO.board.zip)

Do you know what GPIO's are used for the SD Card reader?

closed 2026-11-02 by 0xchocolate